sanjay900/SantrollerConfigurator #1: whammy not working anymore and cant change settings in tilt sensor

Grayhoof, 2021-03-07:
my whammy bar suddenly stopped working... it still works with bluetooth but not with the arduino

also the MPU 6050 orientation setting doesnt save for me for some reason aswell

im using a arduino pro micro 3.3v

sanjay900, 2021-03-08:
are you using the preview build or the normal one?

sanjay900, 2021-03-08:
also what do you mean when you say it works with bluetooth?

sanjay900, 2021-03-08:
oh right your using wii emulation mode right? Does the config tool tell you that it thinks a wii guitar is connected?

Grayhoof, 2021-03-08:
1. i used both preview and normal build even an older version it just suddenly stopped working

2. yes i use wii adaptor mode everything works great even the tilt sensor but only the whammy just stopped working for some reason

sanjay900, 2021-03-08:
Could you try swapping to direct mode, inverting the whammy, and then
swapping back to wii? I think I might have broken this by accident while
making some other changes. If this works then I'll implement some things
differently for a proper fix

Grayhoof, 2021-03-08:
hmm i gave it a try but it didnt work sadly

Grayhoof, 2021-03-09:
okay so i tried arround in the software abit,., the whammy works when i configure the guitar as a xinput gamepad instead of a guitar.. it sees it as x rotation then i dont know if that helps

sanjay900, 2021-03-10:
okay in that case i have an idea as to what might be doing this. let me test some things and ill get back to you

sanjay900, 2021-03-11:
fixed with https://github.com/sanjay900/guitar-configurator/commit/c44daed88b5c99aec1ca42463d5865800e30e3e4

This is currently only fixed on the preview build, which you can get from the following link:
https://github.com/sanjay900/guitar-configurator/releases/tag/preview

closed 2021-03-11 by sanjay900