sanjay900/SantrollerConfigurator #2: Trouble with Link to Clone Hero menu - unknown version

NickBonet, 2021-03-09:
I'm having an issue with the "Link to Clone Hero" menu, I've pointed to my game install directory in the menu, but it still reports unknown version and thus can't launch from there/have my APA102 LEDs switch to star power/open note mode. 

[image]

I'm running v8.0,2 which was released a few days ago.

NickBonet, 2021-03-09:
My Clone Hero directory: 
[image]

sanjay900, 2021-03-10:
What version of the game are you using?

sanjay900, 2021-03-10:
ah yeah, its been a while since ive had the time to update the LED bindings, ill try and get to doing that this weekend

NickBonet, 2021-03-10:
Ah that makes sense then, thanks!

sanjay900, 2021-03-11:
Should be fixed with https://github.com/sanjay900/guitar-configurator/commit/bc2884599a06bb2338524abd8dff9a70db9b8cb5
turns out i did implement that version, but the files i was grabbing from the clone hero site to detect versions just was wrong for windows and that broke things. Ive changed it so you can just pick a version now instead of it trying to detect.
https://github.com/sanjay900/guitar-configurator/releases/tag/preview

closed 2021-03-11 by sanjay900