sanjay900/SantrollerConfigurator #4: Arduino UNO based guitar works outside of CH, but not in game.

NickBonet, 2021-03-12:
Running v8.0.2 with an UNO based guitar, and for some reason no inputs register within Clone Hero.
Dropped back to 8.0.1 to test, everything appears to work normally on the board

sanjay900, 2021-03-13:
What about the preview build?

NickBonet, 2021-03-13:
I actually tried both 8.0.2 and preview build, since I used preview for the other fixes you put in, and re-flashed the UNO each time for testing purposes, but it would not work within Clone Hero. It worked fine checking USB game controller settings in Windows and checking button inputs though

sanjay900, 2021-03-14:
Is clone hero binding it correctly? Does it all show up in the controller menu when you hit spacebar?

NickBonet, 2021-03-16:
I would have to check again next time I disassemble the guitar, currently in the process of moving so not able to right at the moment. I remember it at least showing up as a guitar in game, don't think I saw controls actually bound though.

sanjay900, 2021-03-21:
What operating system are you using also? I could see this happening on Linux

sanjay900, 2021-03-22:
The latest build may have done something, it contains a bunch of fixes specifically for the UNO

sanjay900, 2021-03-27:
Also, it was pointed out to me that leds weren't working on windows so that's been fixed now

nikorev, 2021-03-29:
Can confirm pro micro flashed with xinput does not get detected in Ubuntu (works on Windows, haven't tested OSX).

Flashing to PS guitar works as an alternative, but would much rather prefer xinput mapping.

sanjay900, 2021-03-31:
Update your kernel, they only started supporting my controllers recently

nikorev, 2021-04-02:
Thanks @sanjay900 can confirm the controller works on Ubuntu 20.04's new v5 kernel vs Ubuntu 18.04's v4 kernel.

closed 2022-04-25 by sanjay900