sanjay900/SantrollerConfigurator #5: Arduino Micro stuck on "Waiting for Avrdude"

mnkhouri, 2021-03-14:
When attempting to program my Arduino Micro, the progress bar was stuck at 0% saying "waiting for Avrdude".

To fix it, I needed to:
1. Reset the Arduino twice rapidly to put into "bootloader mode"
2. Now, in the Guitar Configurator UI, the device appears with "DFU mode" somewhere in the name
3. Rapidly click next to program the device (you only have ~5 seconds before the chip exits "bootloader mode")

@sanjay900 helped me figure this out -- thanks!

I'm writing this issue so that if someone else is stuck on "waiting for Avrdude", they'll maybe find these instruction. If it's possible, perhaps it's worth adding a message to this effect or detecting this situation within the program?

sanjay900, 2021-03-14:
It's interesting, cause it really should have done this process for you, but I will add something about it to the guide, as some arduinos can be a bit odd

FalconerGH, 2021-06-15:
Hi, I am having this same problem.

I have tried pressing the reset button twice rapidly but nothing happens.
I'm not really experienced with arduino and have looked online but to no avail.
Can you help me?

FalconerGH, 2021-06-15:
I also just realised that my arduino is nano and not micro. There is no option for nano on the list, does this mean I can't use it for my guitar?

sanjay900, 2021-06-15:
Yeah you can't use nanos for this as they don't support full usb emulation

closed 2022-02-02 by sanjay900