sanjay900/SantrollerConfigurator #6: Pcb switch holder les paul

BrendanSiche, 2021-04-12:
Hello,

I hope this isn't too off-topic, I'm currently doing the conversion for my wii LP. In the tutorial you have a 3d printed plate to hold the strum bar switch, but I can't find the ressources anywhere to print it (I found some WT guitar, a 3D print replacement for the original switch, and the fretboard/frets, but can't find the cherry mx strum bracket for LP. Do you remember where you got your files/did you ahve to make it yourself ?

Thanks !

sanjay900, 2021-04-13:
Unfortunately that wasn't mine, I was sent the picture and figured it would look nice in the tutorial

BrendanSiche, 2021-04-13:
Okay, no problems !
Thanks again for the tutorial !

closed 2021-05-09 by sanjay900