sanjay900/SantrollerConfigurator #8: Experiencing erratic analog stick behavior on 8.4.7

NickBonet, 2021-04-13:
Just upgraded to 8.4.7, and I noticed my analog stick started behaving very erratically. I tried the new calibration controls but even then the analog would only register some inputs and not others. I downgraded back to 8.,0.1 and reflashed my controller and the analog stick works perfectly again for reference.

sanjay900, 2021-04-13:
Is this with with wii mode and a wiitar?

NickBonet, 2021-04-13:
This is with a Wii LP, XInput Direct mode.

Get Outlook for Android

sanjay900, 2021-04-13:
Are you using it as an adaptor or direct wired

NickBonet, 2021-04-13:
It is directly wired.

sanjay900, 2021-05-13:
when you calibrated it, how did you calibrate it? whats interesting is i have a guitar here where i have a analog stick bound, and after calibration it seems to work perfectly

closed 2022-04-25 by sanjay900