sanjay900/SantrollerConfigurator #9: Configurator freezes / won't launch when configured Uno is connected

JRandomHacker, 2021-07-05:
I'm attempting to set up an RF guitar using Uno as a receiver and Sparkfun Pro Mini as a transmitter. I successfully programmed the Uno, moved to configuration, selected "Enable RF", and then wrote the configuration to the Uno.

Result:
Ardwiino is no longer visible in Guitar Configurator / Configurator locks up
Ardwiino is visible as a Game Controller under Windows
Opening Configurator and plugging in Ardwiino causes Configurator to lock up
Opening Configurator when Ardwiino is already plugged in causes Configurator to fail to open (process starts, no UI)

sanjay900, 2021-07-07:
Have you got the rf transmitter set up yet?

sanjay900, 2021-07-07:
The config is stored on the transmitter, so it will probably lock up if it
isn't able to talk to it

JRandomHacker, 2021-07-08:
Alright, I've tried again, starting from burning fresh bootloaders and the Blink sketch on both boards.

1 - program Pro Micro
2 - Toggle "Enable RF" and press "Program RF Transmitter"
3 - Arduino Pro Mini shows up on correct COM port as "Arduino Pro Micro" and only supports configuring, not programming.

sanjay900, 2021-07-11:
are you able to verify that the RF stuff works with an  rf testing sketch?

sanjay900, 2021-07-12:
also, i see that you mention pro micro and pro  mini a bunch, is it a micro or a mini they are completly different boards

JRandomHacker, 2021-07-13:
I'm using a Pro Mini 3.3v for transmitter, and a Pro Micro 5v for receiver.

sanjay900, 2021-07-15:
ah okay, and what nrf24l01 board are you using?

JRandomHacker, 2021-07-22:
I'm using the [Sparkfun Transceiver Breakout](https://www.sparkfun.com/products/691) - the power input is listed as 3.3v - 7v and the data pins are all 3.3v and 5v-compatible

sanjay900, 2021-07-22:
Alright, see if you can get the transmitter working with a demo sketch.

sanjay900, 2022-04-06:
try the latest build out, there were some issues with the config on the uno

closed 2022-04-25 by sanjay900