sanjay900/SantrollerConfigurator #10: Won't recognize Wii guitar

oboewan42, 2021-09-17:
I wired up a Wii adapter using a 5V Pro Micro clone, a voltage regulator, and a nunchuk breakout. Everything is wired correctly and I can access the guitar just fine via the NintendoExtensionCtrl library.

However, when I try setting up the adapter using the configurator, then open the Windows controller tester, it doesn't respond to any inputs. The top left corner of the configurator window says "Ardwiino/Arduino Pro Micro (5V)/XInput Guitar Hero Guitar/Wii Unknown Extension".

sanjay900, 2021-09-18:
Odd, I presume you've tried unplugging and replugging after programming?

oboewan42, 2021-09-18:
Yeah, both on the guitar end and on the USB end.

oboewan42, 2021-09-18:
Did some more testing. Tried adding a 6050 for tilt, but it wouldn't recognize it either.
Then, I tried changing to direct mode - it'll detect the pins, but not the 6050. 
Using the Arduino IDE I can read the 6050 just fine.

Conclusion - it's probably something to do with the i2c bus in general.

sanjay900, 2021-09-19:
Odd, it used to work just fine so I'm not really sure what's happened there, do you want to try an older build and see if one of those works?

oboewan42, 2021-09-19:
Looks like it broke between 8.4.7 and 8.5.1

oboewan42, 2021-09-29:
Also, this is apparently unrelated, but tilt output is really erratic if I have "Invert" checked, and there's no option to calibrate, if I'm using a 6050.

sanjay900, 2021-09-29:
are you using the right axis?

oboewan42, 2021-10-04:
Yeah, it looks like the issue is that if I have it set to "inverted", it'll blink between the inverted and non-inverted value. As a workaround I just mounted my 6050 upside down.

oboewan42, 2021-10-04:
Switched to a 3.3V Pro Micro, updated back to 8.5.4, and that appears to have fixed both the main issue and the inverted-axis issue. However, there's another issue here: the tilt sensor only shows up in joy.cpl, not in Clone Hero.

oboewan42, 2021-10-05:
I also noticed that, if I open up joy.cpl's calibration tool and check "Display Raw Data", the range for the tilt axis is different based on which Pro Micro I'm using. (On the 3.3V model, it ranges from 0-63, while on the 5V model, it ranges from 0-65535.)

NickBonet, 2022-01-05:
I experienced similar issues with Wii guitars not being recognized on 8.5.4, both on an adapter I just built and one of my adapters I updated for testing purposes. At first I thought the build went bad but noticed the same behavior on my old adapter that updated within 8.5.4.

sanjay900, 2022-01-08:
Yeah, I'm in the process of rewriting everything so I don't really want to
debug this issue

sanjay900, 2022-03-03:
i have a fix on this old firmware but am waiting for some azure fixes before i can push out a preview build, ill have one out tomorrow.

closed 2022-03-03 by sanjay900

sanjay900, 2022-03-04:
https://github.com/sanjay900/guitar-configurator/releases/tag/preview