sanjay900/SantrollerConfigurator #12: Writing the Raspberry Pi Pico as a PS3 Guitar Hero guitar causes issues.

silverrat23, 2022-01-03:
In the process of writing to the Pi Pico after hooking up all the buttons and whammy.  

Clicking "Write" would freeze the Guitar Configurator program, cause the Windows 10 PC to go into a state of changing monitor settings (screen blacks out intermittently) and will not recognise as a device at all. Did managed to revert the changes by quickly changing the settings back in the program before the Pico was fully read after re-inserting it (now using Direct, XInput, Guitar Hero Guitar). I'm scared to re-create the issue for a thorough explanation, as the issues are not consistent. Here are the settings I used:

Controller Connectivity Type = Direct
Controller Input Type = PS3, Guitar Hero Guitar
All frets, strum, Dpad and whammy were connected. 
Currently using v8.5.4

How can this be fixed?
Thank you.

sanjay900, 2022-01-03:
Yeah I think PS3 isn't quite working on Picos at the moment

silverrat23, 2022-01-04:
So far, I presume the Pico's only been tested as "Direct, XInput, Guitar Hero Guitar"? Will there be further development to its compatibility?

sanjay900, 2022-01-05:
at the moment the current pico implementation is a bit of a hack, i'm in the process of rewriting the entire thing as the original code really wasn't designed to support multiple microcontrollers like this. It's taking some time to do that though which is why it probably looks like this project is dead

silverrat23, 2022-01-05:
> at the moment the current pico implementation is a bit of a hack,
Fair enough, the Pico is rather new compared to Arduinos. 

> i'm in the process of rewriting the entire thing as the original code really wasn't designed to support multiple microcontrollers like this. It's taking some time to do that though which is why it probably looks like this project is dead

Don't stress, what's already there is amazing. It'll come out when it's ready. 
Keep up the awesome work, can't wait to hear more about this.
Thank you so much!

sanjay900, 2022-03-03:
i have fixes for this but will have to wait until tomrrow as im having issues with azure at the moment

closed 2022-03-03 by sanjay900

sanjay900, 2022-03-04:
https://github.com/sanjay900/guitar-configurator/releases/tag/preview

sanjay900, 2022-03-04:
an 8.5.5 will be out shortly with this fix

sanjay900, 2022-03-04:
https://github.com/sanjay900/guitar-configurator/releases/tag/v8.5.5