sanjay900/SantrollerConfigurator #13: Rockband drums

M0RRiS0on, 2022-01-09:
Rockband drums cymbals undefined
Can't bind them to pins
[image]
In list view mode they are also absent
[image]

M0RRiS0on, 2022-01-09:
If this program emulates ps3 drums maybe adding cymbals functionality possible by combo (DrumColor+modificator) for them. According to this forum: 
https://dwsk.proboards.com/thread/2258/playstation-4-rb4-pro-cymbals
About drums before ps4 version:
"Before, a tom was one button, and that same colored cymbal was that button + a shoulder bumper button, for example."

sanjay900, 2022-01-11:
Ah yeah cymbals were one of those things I never added support for as the
code just wasn't designed for it at the time, I may eventually add it though

M0RRiS0on, 2022-01-13:
Ok, hope my info can help somehow.  Will be waiting eagerly new updates!

sanjay900, 2023-01-09:
Ive implemented drums in the new firmware that's being worked on with full cymbal support and all that, but there's still a lot more to implement before I can release that.

closed 2023-01-09 by sanjay900