sanjay900/SantrollerConfigurator #14: issue with arduino pro micro

tekman123, 2022-02-02:
so, when I plug the pro micro in and start up the configurator I get the same issue as one of the other people in here, it starts avrdude and never moves past 0.00 so I did the trick where you short the reset and gnd terminals twice to throw it into bootloader mode. Well, when I do that it brings up another pro micro in bootloader mode on Com 7 but it will not let me program this one.
[image]

I tried several time just to make sure it just wasn't me failing

sanjay900, 2022-02-02:
If your getting this error when doing the bootloader mode trick, it means you aren't being fast enough. Annoyingly, the bootloader stuff only shows up for  8 seconds or so.

closed 2022-02-02 by sanjay900

hex337, 2022-11-27:
I have been trying to connect and program an RP2040 from adafruit, and I'm getting stuck on waiting for Avrdude and it's sitting at 0%. Tried to do it within 8 seconds of connecting like the other thread, but still no dice.

sanjay900, 2022-11-27:
Read the guide, picos dont use this method at all, you need to hold bootsel as you plug them in.

hex337, 2022-11-27:
Yeah, I held the bootsel button, but it still is stuck on this with no output on the bottom. I did change the USB driver on it to USB Serial which I now think is even more messed up :(

sanjay900, 2022-11-27:
Swap the driver back, it should just be a usb drive, and at that point the tool should pick it up as a pi pico and nothing else.

hex337, 2022-11-27:
Friend request sent, thanks!

hex337, 2022-11-27:
Ok, to update this thread, it looks like the issue I had was that my PC was not recognizing my RP2040 as a usb drive. If that doesn't happen, then it tries to program it like an arduino which won't work.