sanjay900/SantrollerConfigurator #16: Pro Micro 5V takes approx 40 seconds to detect in Windows 10 when configured as Xinput Guitar Hero Guitar - Direct

puredave, 2022-03-01:
Using latest version and configuring as a Direct Xinput Guitar Hero Guitar controller. Whenever the pro micro (5V) is connected to USB it will take about 40 seconds to finish adding the controller to Windows. It also seems to hang the entire system during this time. This occurs on a Windows 10 Dell Laptop (Intel CPU)  and my main Clone Hero PC (AMD Ryzen 1600 also Windows10). 

I tested it with 2 of my Windows 11 PCs and the issue is non-existent and there is no problem. 

If I configure it as a Direct PS3 Guitar Hero Guitar controller this issue does not occur on Windows 10 and it connects to USB and is ready to go instantly. All well and good but I don't particularly want to configure it as a PS3 controller ;)

Tried with version 8.4.5 and same issue.

Is this a known issue with Pro Micro 5V & Windows 10 (when configured as a Direct - Xinput Guitar) and is there a work around/bug fix or is there something strange afoot at the circle-K?

quanpunchhhhh, 2022-03-03:
Happens with me too, but I'm using an Adafruit ItsyBitsy 3v. Tried on two computers as well, one a windows 10 pc and the other a windows 10 laptop. It freezes up my mouse and keyboard, but not the display for almost exactly 42 seconds before my computer responds again

https://user-images.githubusercontent.com/49852204/156497837-f1658933-5620-47bc-9d09-f8ad3e949b29.mp4

.

sanjay900, 2022-03-03:
I really have no idea at this point, and my intention is to make ps3 the default anyways, as clone hero can poll ps3 mode with less cpu overhead. Im rewriting the entire thing anyways so this problem may solve itself. Any specific reason you dont want to use ps3 mode?

sanjay900, 2022-03-03:
its a pain because this used to be a problem and then it was fixed for a while, so i haven't got a clue why the problem has reappeared

sanjay900, 2022-03-03:
And i do suspect half the problems are due to some design decisions that just aren't in the new firmware, but it is quite a while away

sanjay900, 2022-03-03:
one thing worth trying though, disable the "map start + select to home" option

puredave, 2022-03-03:
> I really have no idea at this point, and my intention is to make ps3 the default anyways, as clone hero can poll ps3 mode with less cpu overhead. Im rewriting the entire thing anyways so this problem may solve itself. Any specific reason you dont want to use ps3 mode?

I assumed xinput was the best for direct? Incorrectly it would seem!

puredave, 2022-03-03:
> one thing worth trying though, disable the "map start + select to home" option, some people have mad issues with it

Yeah i have tried with that enabled and disabled and it made no difference. Weird how windows 11 does not have this problem.

Are you aware of other windows versions with this same issue or is it just 10?

sanjay900, 2022-03-03:
Oddly I test with win 10 and have no issues

sanjay900, 2022-05-07:
have any of the recent firmware changes changed this in any way or is it still happening?

puredave, 2022-05-10:
> have any of the recent firmware changes changed this in any way or is it still happening?

Tested with latest version. Issue still persists on windows 10. No problem if connected to Windows 11 PC or if configured as a PS3 controller.

I am starting to wonder if it is something to do with the clone Pro Micro, perhaps some variance in the components from the knock-off vendors so it may or may not happen depending on the boards. All my boards are from same source so all exhibit the same issue.

sanjay900, 2022-05-11:
nah, i think microsofts windows 10 driver for xbox controllers is just particularly  buggy, i remember people had issues with crashes even with normal xplorers and other third party controllers. I suspect the rewrite might help a little though, as that will make the controllers be a lot closer to a normal controller, cause at the moment the controller i emulate is a tad stripped down.

quanpunchhhhh, 2022-06-19:
Update: this does not happen with my picos, only my 32u4 based boards

sanjay900, 2022-06-19:
that makes even less sense haha, but interesting

puredave, 2022-10-11:
Will be able to check it out during the week and get back to you

sanjay900, 2023-04-13:
Im hoping whatever caused this will be fixed by the new firmware (if its even still a bug) since the new xinput implementation is a lot closer to a real xinput controller.