sanjay900/SantrollerConfigurator #19: Green fret

0Peligr0, 2022-04-20:
The green fret doesnt work, what can I do?

sanjay900, 2022-04-21:
How have you wired things, what arduino is it? Usually, a single fret not working is a wiring issue

0Peligr0, 2022-04-21:
> How have you wired things, what arduino is it? Usually, a single fret not working is a wiring issue

I think all the cables to the arduino are well connected because all the other buttons work.
Maybe is a problem of the connections of the guitar, but where can I find the blueprints of the guitar inner connections?

sanjay900, 2022-04-21:
Oh wait is this using the Arduino as a wiitars adapter?

sanjay900, 2022-04-21:
The pcb for the frets is pretty easy to understand just by looking, I don't think any blueprints exist

0Peligr0, 2022-04-22:
> The pcb for the frets is pretty easy to understand just by looking, I don't think any blueprints exist

It is easy but there is a black dot on the middle of the board and I cant see where the cables are going through. There is any way to chek the frets?

sanjay900, 2022-04-22:
Oh wait what guitar is this?

0Peligr0, 2022-04-23:
> Oh wait what guitar is this?
> […](#)

Is the guitar hero 2 guitar the red one for the ps2

sanjay900, 2022-04-23:
You can always work out the frets with a multimeter in continuity mode. It will beep when the two contacts are connected together, and you can use that to work out which pin is which. Could you send a photo of the fret pcb?

0Peligr0, 2022-04-23:
> You can always work out the frets with a multimeter in continuity mode. It will beep when the two contacts are connected together, and you can use that to work out which pin is which. Could you send a photo of the fret pcb?

[image]

sanjay900, 2022-04-24:
does the guitar work okay on a console?

sanjay900, 2022-04-24:
it looks like with that pcb, the pin all the way on the left is ground (the one with the thicker trace on the fret pcb), and green will be one of the three on the side you circled. should be easy enough to test between ground and the other pins to work out which one is green, and to work out if something is wrong with the fret itself

sanjay900, 2022-04-24:
if you have the config tool open, what does it say in the top left corner?

sanjay900, 2022-04-24:
Ah actually i might have just found a issue with my handling of PS2 controllers, can you try out the latest version of the tool for me? I could see it being the issue with your green fret not working.

0Peligr0, 2022-04-24:
> Ah actually i might have just found a issue with my handling of PS2 controllers, can you try out the latest version of the tool for me? I could see it being the issue with your green fret not working. Theres a button to update inside the config tool, atm the latest is 8.6.2.
> 
> https://github.com/sanjay900/guitar-configurator/releases/tag/v8.6.2

I just tried the new version and nothing, and in the left corner say 
[image]

sanjay900, 2022-04-24:
that image does not seem to have loaded, can you try uploading it again

0Peligr0, 2022-04-24:
[image]

0Peligr0, 2022-04-24:
> that image does not seem to have loaded, can you try uploading it again

[image]

sanjay900, 2022-04-24:
ah, that unknown isnt helping us much! to confirm, your using this to make a ps2 controller adaptor right?

0Peligr0, 2022-04-24:
> ah, that unknown isnt helping us much! to confirm, your using this to make a ps2 controller adaptor right?

Yes

sanjay900, 2022-04-24:
i presume whammy and tilt are also not working correct?

0Peligr0, 2022-04-24:
> i presume whammy and tilt are also not working correct?

They work well

sanjay900, 2022-04-24:
ah okay, then this isnt the problem i thought it was

sanjay900, 2022-04-24:
did tilt show up as a button or as a axis?

0Peligr0, 2022-04-24:
> did tilt show up as a button or as a axis?

No, the programs dont recognize it

sanjay900, 2022-04-24:
but you just said that they work well?

0Peligr0, 2022-04-24:
> but you just said that they work well?

Axis, sorry

sanjay900, 2022-04-24:
and whammy shows up as an axis too and works?

0Peligr0, 2022-04-24:
> and whammy shows up as an axis too and works?

Yes

sanjay900, 2022-04-24:
interesting

sanjay900, 2022-04-24:
> it looks like with that pcb, the pin all the way on the left is ground (the one with the thicker trace on the fret pcb), and green will be one of the three on the side you circled. should be easy [...]

are you able to confirm the green fret is working with a multimeter?

0Peligr0, 2022-04-24:
I dont have a multimeter yet but Im trying to get one this week

sanjay900, 2022-04-24:
ah okay, can you try the latest version (8.6.5) and let me know what the text in the top corner says now?

0Peligr0, 2022-04-24:
> ah okay, can you try the latest version (8.6.5) and let me know what the text in the top corner says now?

It sais the same

sanjay900, 2022-04-24:
hmm, can you try unplugging and replugging the arduino from the computer

sanjay900, 2022-04-24:
it really should not say Unknown, it should say guitar hero guitar or something

0Peligr0, 2022-04-24:
> it really should not say Unknown, it should say guitar hero guitar or something

Now it say PlayStation No Device

sanjay900, 2022-04-25:
okay, that is different, are the other buttons still working?

sanjay900, 2022-04-25:
do you have a discord? would be far easier to converse on that vs github, im sanjay900#5549, if not no problem

sanjay900, 2022-04-25:
Can you try the latest version

0Peligr0, 2022-04-26:
> okay, that is different, are the other buttons still working?

yes, the other buttons sill working, I've tested the green fret first contact and it mark well like the others, but i dont know how to know if the resistors work, how can I measure it with the multimeter?

closed 2022-05-02 by sanjay900