sanjay900/SantrollerConfigurator #21: [Help]: Unable to run this on OS X

hex337, 2022-04-28:
### Please describe your issue.

The README says that this should work on mac OS X, but it looks like the releases just have a windows exe and linux app package. Wasn't sure if this was still supported, or if there's something I'm missing!

### Version

yes

### What operating system are you using?

Mac OS

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2022-04-29:
ah yeah, at some point the build for osx stopped working and i just never got around to figuring out why, though tbh i could never do proper signing

sanjay900, 2022-04-29:
I don't really want to purchase a apple developer account, and without one i dont see a point in releasing a macos app.

sanjay900, 2022-04-29:
You can however configure a guitar for PS3 mode on a windows or linux machine, and then it will work on a mac.

sanjay900, 2022-04-29:
there is now a dmg on the latest build.