sanjay900/SantrollerConfigurator #22: [Request] Bind an existing button to simultaneously act as tilt activation

muskit, 2022-05-07:
For those without a tilt sensor, being able to bind tilt activation to a button (like Select) would help with games that do not support activating Star Power by button (like GH3 PS3).

sanjay900, 2022-05-07:
you should be able to just select "digital" as your tilt sensor

sanjay900, 2022-05-07:
ah right but the pin is in use

sanjay900, 2022-05-07:
https://github.com/sanjay900/guitar-configurator/releases/tag/v8.7.6

sanjay900, 2022-05-07:
should be able to bind two things to the same input now, the dialog has an option for keeping both.

muskit, 2022-05-09:
My bad, I forgot to mention an important thing: I'm not using direct connectivity, I'm using a Wii guitar. What I want is a way to bind the Select button on my WiiTar to also act as a tilt activator.

sanjay900, 2022-05-09:
ah thats quite a different request

sanjay900, 2022-06-20:
This sort of thing will be possible in the new version of the tool that is being worked on, however, it is quite a while away at the moment

sanjay900, 2023-03-22:
Can confirm, the new firmware has this feature coded in, you can essentially map whatever to whatever, including reusing inputs and all sorts of things. Theres also built in digital to analog and analog to digital as well, so you can easily map tilt to buttons or buttons to tilt and all that sorta stuff.

muskit, 2023-03-25:
I may have worded my request originally quite badly. I would like to bind an **input from my Wii guitar** to activate tilt, not a button that's wired to my board.

sanjay900, 2023-03-25:
Nah when I say new firmware - there's a new one in development that's a total rewrite, and this is supported in that. Not released yet though.

muskit, 2023-03-25:
Ohh got it. Sorry for the misunderstanding!

sanjay900, 2023-03-25:
When I originally wrote that comment I didn't intend to ever release any more versions of the current tool hence the confusion

sanjay900, 2023-03-25:
[image]