sanjay900/SantrollerConfigurator #23: [Help]: Configurator Crashes After Flashing New Firmware

muskit, 2022-05-07:
### Please describe your issue.

After updating the configurator and flashing updated firmware onto my Pro Micro, the Configurator crashes after clicking on "Start Configuring."
I also noticed that after firmware is flashed the top left status still displays "Outdated, Continue to update."

Running on v8.7.6, updated from v8.7.4 using the update prompt upon starting up.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 3.3V

### How is your controller programmed?

Wii

sanjay900, 2022-05-07:
seems like when i redid the build scripts to add in all the various pi pico versions, something went wrong and the arduino builds dont have the correct version numbers anymore, im having a look now

sanjay900, 2022-05-07:
there is a v8.7.7 compiling now that will fix this

sanjay900, 2022-05-07:
give it 10m or so to compile and release

sanjay900, 2022-05-07:
resolved by 8.7.7

closed 2022-05-07 by sanjay900