sanjay900/SantrollerConfigurator #24: [Request]: RF Setup with Pico

RE-Gaming, 2022-05-10:
### Please describe your request.

Is RF TX and RX setup with Pico on the roadmap?

The setup is for Wii Guitar input --> PS3 output

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2022-05-10:
Theres currently a rewrite on the roadmap, as the current RF code isn't great. Part of that rewrite will be a total rewrite of the RF code, and that will give it pico compatibility as well. This is a while away though.

RE-Gaming, 2022-05-10:
No worries! The wired implementation currently works amazing. I've been able to get this to work with Rockband 4 on PS4.

sanjay900, 2022-05-14:
Honestly I haven't tested this at all but if your lucky you might find that using the pins from here may just work 
https://github.com/sanjay900/Ardwiino/blob/master/src/pico/pins_arduino.h

RE-Gaming, 2022-05-18:
I tested out the pinout you linked above but didn't get quite far enough to be able to test it.

When I program the receiver Pico, I set it up like I normally would the wired version. When I click program RF --> program transmitter I am able to select a second Pico but the program hangs after I choose to program the transmitter Pico.
[image]

The configurator hangs at "Waiting for Reconnection"

sanjay900, 2022-05-18:
ah okay

RE-Gaming, 2023-01-29:
Hello,
Any update on the timelines for this feature add?
Thanks!

sanjay900, 2023-01-30:
none at this time

sanjay900, 2023-02-13:
Finally actually getting around to writing something, so I will be able to see if I can get something that is just always stable

sanjay900, 2023-04-03:
So a few updates
I finally got around to properly implementing RF yesterday, including the UI elements for it
[image]
[image]

But for the pico specifically, i intend to also support something with the pico w. My intent is to support using Bluetooth, so you can build just the transmitter for PC use, but for console you you can then build a bluetooth receiver as well. I have the transmitter side of the bluetooth equation sorted and just need to write the receiver which i don't expect to take me long.

sanjay900, 2023-04-03:
You can probably see why i was so hesistant to give you a timeline though, this stuff happens at a snails pace as i just put my free time into it

sanjay900, 2023-06-29:
Yeah okay, i will be dropping RF support, its just such a mission to keep that working.

closed 2023-06-29 by sanjay900