sanjay900/SantrollerConfigurator #25: Digital Tilt Activation is now broken

puredave, 2022-05-12:
v8.8.3

Setting up digital tilt no longer works correctly. It will configure fine in the tool, but in-game or in the windows USB game controller tool. It no longer works correctly and behaves as if the guitar is being tilted even when it is not. (maybe the value in inverted?) 

v8.5.9

Works as intended.

Tested with both Pro Micro 5v & Pi Pico boards, same issue with either. Configured as Direct - Xinput

-----

On a side note, when I configure guitars as Direct - PS3 controllers, digital tilt doesn't work at all. It is fine in the configurator but then no response in-game or in windows USB game controller tool. Doesn't work with Pi Pico or Pro Micro 5v. This is with v8.5.9 and 8.8.3.

sanjay900, 2022-05-13:
yeah tilt is pretty bugged on ps3 controllers, i can guess how i broke tilt ill revert that change

sanjay900, 2022-05-13:
Fix is out

closed 2022-05-13 by sanjay900