sanjay900/SantrollerConfigurator #26: [Request]: I have a solution for USB Drums (Wii or Xbox) into the PS3/PS4 mode, is it worth incorporating?

RE-Gaming, 2022-05-14:
### Please describe your request.

I worked out a solution to get USB inputs from xbox drums (tested) or wii drums (no reason why they shouldn't work too) and get them to work on the PS3/PS4. I wanted to share this as it could really help expand the available supported instruments as inputs. This also allow people to use the drum without having to tear them apart to. 

So my solution uses an RPI Zero to host the Xbox/Wii USB drum set, uses Pygame to poll the drum inputs and then sets GPIO LOW/HIGH accordingly. These GPIO are then tied into your existing Pico code using the Direct Wire mode --> PS3 Drum set.

This USB "passthrough" should work with USB dongle guitars as well but that is untested, so TBD on that one. I think the Pico can act as a USB host but I don't know how to do that so I used the RPI Zero instead.

[drums.txt](https://github.com/sanjay900/guitar-configurator/files/8693978/drums.txt)
[image]
 

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-05-15:
You can actually do this with just a pi pico with no additional hardware and that's on the roadmap in the future

closed 2022-05-15 by RE-Gaming

sanjay900, 2023-04-16:
just as a sidenote, I have implemented this into the new firmware in development