sanjay900/SantrollerConfigurator #28: [Help]: Whammy Bar not sending midi

ncint, 2022-05-22:
### Please describe your issue.

Hello! 

I configured everything ( p2 guitar hero) and the whammy bar is not sending **midi**. Everything else is working.

I calibrated the whammy in the guitar configurator and it reads the input so i guess it´s not the potentiometer...
I tried  note send and Control message and no luck.  Any thoughts?

- By the way , Is there any way to send CC messages? (ex CC23)

Thanks for you work!!

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2022-05-23:
the MIDI implementation was pretty half baked, not really surprised it doesnt work right tbh

sanjay900, 2022-05-24:
In theory you can send CC commands, you would just have to work out what their corresponding note is since i don't have that box supporting anything besides notes