sanjay900/SantrollerConfigurator #29: [Help]: no tilt wii guitar

alidan, 2022-06-17:
### Please describe your issue.

Ok, I am following this guide 
https://sanjay900.github.io/guitar-configurator/guides/wii.html
I have no tilt with a MPU 6050

not really sure what I did wrong or if this is a software thing. 

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2022-06-17:
how did you wire it all?

alidan, 2022-06-17:
Ok, I will go through this one at a time and make a picture for what I believe the current wiring is. 

ok, the guide while correct in pins was vague enough that looking at pinouts had me wire it up wrong at first. the guide says pin 18 and 19, but pin 18 and 19 are ground and sda, so I looked for a pinout for the pi and found 19 and 20 were sda and scl, I assumed numbering was weird because it was 1 pin off of being correct.

that doesn't matter because even then, while I color coded the wires, pinout to the pi was wrong even in my wrong because I swapped what was the sda and scl pins, or at least what I thought were them, there are 2 pinouts I can find for the pi, one has 20 as scl, one has 20 as 'do not use' so I 'corrected' the wiring, but nothing. about an hour or so later someone on the clone hero discord looked up the guide and saw pin 18 and 19 meant gp 18 and 19, or pin 24 and 25 so I cut the connection to 19 and 20 and rewired it to 24 and 25 correctly, and I got the controller portion to work correctly.

[image]
 
if you want actual pictures I can send them but its honestly a complete mess mostly due to error on my side along with misunderstanding the guide, I should also note that while it will work, it periodically cuts out and them comes back in 5-10 minutes later, I think the cable is at fault there so I ordered a new one. 

also, on installing the guitar configurator the bit at 86% the vc redistributable failed to install on 2 different version of the installer and 2 laptops, not sure if that caused a problem or not. I will know tomorrow if the cutting out is a cable issue or not

if you want pics of the actual project I can get you them

sanjay900, 2022-06-18:
The redist failing is fine it just means you already have it installed. I'll fix that guide up at some point and make it more clear that it's gp18 and gp19. Since the controller portion is working the rest should work just fine too. 
Have you tried changing the tilt axis and calibrating it?

alidan, 2022-06-18:
when the guitar was picked up by the software, I tried changing its access, x y and z, and inverting it, and going into the controller menu in windows and calibrating it, there was never any change in values.

am I supposed to see changes in values?

I also tried to calibrate it in clone hero and got enough star power to activate it and tilt didn't work there either. 

the new cables should come today and fingers crossed this fixes the computer seeing the guitar issue, maybe this will also fix tilt, i'm not sure, i'll update after it comes. 

now I do have a question, is there any reason why its all hooked up to just 2 pins rather than any of the other ones? and is there any reason its gp 18 and 19 specifically? I mean direct I believe you choose where things go and you assign them. I should add that i'm asking this from a point of view where im comfortable enough with soldering and following guides but have next to 0 knowledge about any aspect beyond that.

sanjay900, 2022-06-18:
everything was designed around the arduinos where there is only one set of I2C pins, and thus it would complicate things to support more pins. Theres a new firmware on the way that would give me the ability to add something like that much easier, but that is quite a while away.

I may hook up a mpu6050 to a pi pico and test that myself tomorrow, i haven't actually tried because again, most of this was designed around the arduino pro micro and the pi pico was added afterwards.

I also try to avoid the mpu6050 myself as i just much prefer basic digital tilt sensors myself, so i don't really test it much.

alidan, 2022-06-18:
ok, the cable came, I checked the connections on the mpu 6050, it everything should be hooked up correctly.

I will say that my mpu 6050 is from https://www.adafruit.com/product/3886#tutorials

someone also linked me a listing where they bought 4 or 5 for about 1$ each, they were a bit different at least in look to what I have. and yea a simple sensor would have definitely done the job but I figured that this would probably have the least chance of being annoying for whatever reason, mostly accidental activations or whatever getting stuck either on or off inside of the basic sensors. 

and thanks for looking into this for me, i'm going to move forward with mounting components and making a backplate for the guitar as the main issues for gameplay are solved now, here's hoping the tilt works down the line

sanjay900, 2022-06-18:
yeah there seems to be a lot of mpu6050 variants out there

sanjay900, 2022-06-18:
There is a new build on the way, at some point the "Calibrate" button broke in the tilt gui, and you need to calibrate with that interface in order for tilt to work

sanjay900, 2022-06-19:
try 8.6.7, and use the "calibrate" button inside of the config tool

alidan, 2022-06-23:
sorry this took a bit, some irl stuff came up I had to deal with. 
for me, nothing really seems to do anything tilt wise. I am willing to assume that I may have a bad chip, I have no real way to know for sure.

sanjay900, 2022-06-23:
change the axis to different axes, then hit write, then try calibrating (with the button in the config tool).
When your calibrating, you should be able to see the value move around if its working right

sanjay900, 2022-06-23:
Oh, you have it wired wrong.
You need to provide power to the VIn pin, not the 3Vo pin. VIn takes between 3-5V, 3Vo is an output pin that gives a 3v output if your using 5v.

alidan, 2022-06-23:
ok, thanks, will try that when I get up, getting late for me. will post back with results then

alidan, 2022-06-24:
that did change what it showed me, instead of maximum being maxed out it had it in the middle, but it didn't show any real signs of change, i'll run though xyz and invert but I don't believe it will have any change. it still lights up, think I killed it by putting 3 volt on its 3vo?

sanjay900, 2022-06-24:
doubt it, dont forget to write after changing the axis before actually calibrating. Swing it wildly and the value should change a little it isnt that sensitive

alidan, 2022-06-24:
ok so i just went through every access normal and inverted, wrote calibrated, wrote and then tried to put it in clone hero and got nowhere. I'll take some pictures and a video later of what i'm doing, don't have access to a decent camera for this at the moment.

sanjay900, 2022-06-24:
when you go into the calibration part, does it do anything when you move the guitar around?

sanjay900, 2022-06-24:
otherwise yeah, maybe youve just fried it somehow, though i find it interesting that it seems to be reading something since as you said its in the middle now and not maxed out

alidan, 2022-06-24:
https://www.mediafire.com/file/m8xc2ht9fbvkuj6/IMG_8150.MOV/file

There is me doing calibration, here is the wiring 
[image]
[image]
[image]

there was another image but it was so bad that I scrapped it, it was just trying to get a better angle for the wiring on the pi side, everything should be coded

red - 3v
sda - white
scl - purple
ground - green

I'm thinking, given it works for you, that either I messed yet another thing up in the soldering process or am overlooking something that should be obvious in software, or the chip is a lemon and im not sure which, given it turns on at all the chip being at fault is hard to say unless I killed it.

sanjay900, 2022-06-24:
baiscally, unless you see the bar move in the calibration in the config tool its not working right. The idea is that you calibrate that and it maps whatever motion you use to the axis, so if it sees no motion there then nothing will work. I really have no idea though, the chip could be broken or the wiring could be the issue.

alidan, 2022-07-09:
Ok, I got another 6050 mpu just so I could verify if I killed the original one. 

I have vcc hooked up to 3v
ground to ground
scl to the same scl of the wiichuck
sda to the same sda as the wii chuck

so everything should be wired the exact way it needs to be, and it's still not working. 
any ideas?

the first one I wired up wrong several times so I just assumed I killed it, the second one however I wired up correctly and it functions exactly the same, I have to be doing something wrong and I have no idea what. 

and just so you can see

https://www.amazon.com/dp/B07RXQGGJX

is what I got this time

and checked several times what goes where before I plugged it in.
[image]

alidan, 2022-07-09:
Ok, I just reread the guide to make sure I did it right, and you talk about gp3 and 4 which are also sda scl, am I suppose to connect it to those ones in stead or am I doing it correctly as in the pic...

sanjay900, 2022-07-09:
GP2 and GP3 are for LEDs, the guide is just telling you to ignore them. the tilt sensor and the wii port should be hooked up to the same pins.

alidan, 2022-07-09:
I'm thinking the same thing at least till I direct mod one of the other guitars.

just to be sure, is there any way that whatever the program does skipped over the the mpu 6050 software bit, or if something like that failed, would a lot of other things have had to fail as well?

sanjay900, 2022-07-09:
I don't think so, if you've set it to mpu mode then it should work in theory.

The problem is that the mpus are rather complicated beasts, they have a firmware that the guitar has to upload and all sorts of other stuff so if something isn't working right then nothing will happen.

That's partly why I don't like em much myself, lots more to go wrong, and polling them takes a bit effort on my end too.

I suspect it's something weird with calibration going on, but I have no idea tbh

sanjay900, 2022-07-28:
At this point, im going to deprecate mpu support as im kinda over supporting the mpu, ive had a lot of people report issues and it just isnt worth it for me to continue supporting it. I never really liked the implementation of it much either and it takes up a lot of space in the firmware which is not something i have a lot of.

closed 2022-07-28 by sanjay900

kaju666, 2022-11-16:
> At this point, im going to deprecate mpu support as im kinda over supporting the mpu, ive had a lot of people report issues and it just isnt worth it for me to continue supporting it. I never really [...]

Hi

Downt want to open new thread.
What tilt sensor are now supported? Cant find it on wiki. Mercury one or some digital?

Thanks in Advance, Awasome work, my guitat is almost complete.
Beer is coming to you :)

sanjay900, 2022-11-16:
Yeah, any digital tilt sensor is fine, mercury or ball or whatever

alidan, 2022-11-17:
at least in electronics digital refers mostly to either software settings or it has very definite settings, in this case a mercury switch or a ball switch is either off, or its on, the way it switches may be analogue but the function is digital.

sanjay900, 2022-11-17:
You don't want the ones with three pins like that, though if they are all you can get i think that one is simple enough that you can make it work, you should be able to just use two of the pins

kaju666, 2022-11-21:
> 

Ok, still little not understanding that :) i need to avoid 3 pin devices. So i should buy simple two mercury switches rather

sanjay900, 2022-11-21:
Yeah, the simple 2 pin switch is easiest.

kaju666, 2022-11-21:
Thank you very much :)