sanjay900/SantrollerConfigurator #32: [Issue]: cannot detect pi pico after firmware write

blu18846, 2022-06-29:
### Please describe your issue.

When on Linux, guitar configurator cannot detect a raspberry pi pico after the firmware is written to it.  This applies to both guitars configured on windows and guitars configured on linux.  Tested both on Zorin and Ubuntu

### Version

yes

### What operating system are you using?

Linux

### os-other

Zorin OS

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2022-06-29:
You probably need a udev rule i suspect. Eventually one will be installed automatically, but for the moment either run with sudo or
create the file `/etc/udev/rules.d/70-ardwiino.rules` with the following content
```
ACTION!="add|change", GOTO="ardwiino_rules_end"

SUBSYSTEM=="gpio", MODE="0660", GROUP="plugdev", TAG+="uaccess"

SUBSYSTEM!="usb|tty|hidraw", GOTO="ardwiino_rules_end"

ATTRS{idVendor}=="1209", ATTRS{idProduct}=="2882", MODE="660", GROUP="plugdev", TAG+="uaccess"

LABEL="ardwiino_rules_end"
```
Then add yourself to plugdev and log out and log back in, you may also need to unplug and replug the guitar.

blu18846, 2022-06-29:
Thanks so much for the prompt response, that solved the issue i was having!

closed 2022-06-29 by blu18846

sanjay900, 2023-04-11:
Just added some more notes here:
https://sanjay900.github.io/guitar-configurator/guides/linux.html
if you plan on using ps3 mode and stuff, might want to follow the updated instructions in here.