sanjay900/SantrollerConfigurator #34: [Issue]: Incorrect paths in windows 10 installation causes drivers to not install (Minor Issue)

vCybah, 2022-07-20:
### Please describe your issue.

_Hi, let me first say I did get the app working by going into the file locations that it was telling me the drivers were and installing them myself, so I have fixed this for myself but I just figured I'd drop a comment here in case anyone has the issue again or if it needs to be fixed then hopefully this helps._

During the installation of the newest version, there is an error, (unless this is just for me in which case my bad lol) when the installer gets to launching the drivers but it fails because the system cannot find the files. When I looked at the file paths I noticed some of the slashes are wrong for where its looking. 

**The 2 errors I got:** [here](https://imgur.com/2U26XpX) and [here](https://imgur.com/Ofltz06)

As you can see, some of the slashes are wrong for windows 10, and at first, I assumed that it was a problem since I wasn't using the default installation location but I re-tried with it intact and got the same errors. The solution was to look at the file paths that were in the errors and I installed the drivers that It needed and clicked ignore in the error prompt and It all seems to have worked.

I think that it's just a small error in the code that can be fixed with the next update so I just wanted to bring attention to it in case nobody is aware of it yet, and until then if anyone installing the program and has these errors can't figure out how to fix it hopefully this helps. If that's not the issue and it was just something on my end then I apologize. 

Hope this is useful, thanks for reading.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2022-07-24:
So the second error is unrelated to the file path and instead is just an error as you already have the redist
The first error is interesting, i suspect the slashes are caused by installing it to another directory as i have never seen that before, and you will get a slightly different error if you already have the UNO drivers installed.

sanjay900, 2022-07-24:
Resolved with 8.8.8

closed 2022-07-24 by sanjay900

sanjay900, 2022-10-11:
Usually the VC redist fails because it's already installed Ive never seen
that failure for installing the drivers liie that before and I did test
that so I wonder when it broke

The new tool that's in development will do things differently and avoid
this altogether