sanjay900/SantrollerConfigurator #35: Several Arduino UNO Programming Issues

CapTightpants, 2022-07-22:
### Please describe your issue.

After writing to the UNO, attempting to open the configurator crashes it entirely. When reopening the application, it says the code on the UNO is out of date.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino uno

### How is your controller programmed?

other issue not controller related or I have not gotten this far

CapTightpants, 2022-07-23:
NOTE: Tested using v8.8.5, works as expected.

sanjay900, 2022-07-24:
Looks like when i added in the code for the GH5 neck, it took up too much space, i have disabled that on the UNO for the moment.

closed 2022-07-24 by sanjay900