sanjay900/SantrollerConfigurator #36: [Issue]: button presses not being recognized on newest firmware

redstone47, 2022-07-29:
### Please describe your issue.

after updating to the newest uno firmware no button presses were registering

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino uno

### How is your controller programmed?

Direct

sanjay900, 2022-07-29:
odd, do things like pin detection work at all?

redstone47, 2022-07-29:
yes pin detection works

sanjay900, 2022-08-13:
I suspect this has something to do with the fact that the firmware got bigger recently, and the uno must not quite have the ram or flash space free. Im in the process of rewriting every firmware from scratch, so that should fix this issue once and for all as the new firmware is much smaller

redstone47, 2022-08-15:
i tried to update again today because the whammy doesn't work but it still crashes guitar configurator and no button inputs are recognized

sanjay900, 2022-08-30:
Fixed with 8.9.2

closed 2022-08-30 by sanjay900