sanjay900/SantrollerConfigurator #38: [Issue]: guitar crashing newer versions of clone hero

redstone47, 2022-08-07:
### Please describe your issue.

when guitar is plugged in ch versions 1.0.0.3300 and newer crash

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2022-08-07:
try ps3 rb mode

redstone47, 2022-08-07:
still crashes

sanjay900, 2022-08-08:
Thats pretty odd then since at that point the controller isn't special in comparison to any other controller out there.

redstone47, 2022-08-08:
i tried to reset the firmware back to the normal pi pico one using guitar configurator and it would say it was finished but would still show up as a controller

sanjay900, 2022-08-08:
You can just hold the bootsel button while plugging it into the of then anything will be able to program it again

redstone47, 2022-08-09:
i did that and i still cant reset it

sanjay900, 2022-08-09:
then something is wrong with your pico, as that mode is how they are programmed.
It should show up as a USB drive if you have done it correctly.

sanjay900, 2022-08-09:
you did click save after writing PS3 RB mode right?

sanjay900, 2022-08-09:
out of curiousity, if you have a spare pi pico, does it happen if you just program a pico with nothing hooked up to it?
would be interesting to work out if it happens with a new pico

redstone47, 2022-08-09:
i did save after writing and i do have another pico

redstone47, 2022-08-09:
it still isnt writing but i will try a different computer

redstone47, 2022-08-11:
i dont know how but the issue fixed itself the only problem i can think of is that i somehow had like 7 different versions of guitar configurator installed

sanjay900, 2022-08-12:
good to hear

closed 2022-08-12 by sanjay900