sanjay900/SantrollerConfigurator #39: [Issue]: Connecting Attention to pin 10

ojhernandez, 2022-08-07:
### Please describe your issue.

Hello,

My wife and I are working on building the PS2 adapter with a Raspberry Pi Pico and had a small point of confusion.

On Step 1, when it says "Connect Attention to pin 10" is this referring to GP10 on the Pico which would be the 14th pin or would it be the 10th pin which is also GP7.

We were somewhat confused because this would result in the Attention and Acknowledge pin being connected since Acknowledge is also GP7.

Any clarification would be greatly appreciated!

Thanks!

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2022-08-08:
Everything is in reference to GP pins, I guess I must have missed updating the ps2 guide when I went around and put GP in front of most pin numbers

sanjay900, 2022-08-09:
i have updated the guide

closed 2022-08-09 by sanjay900