sanjay900/SantrollerConfigurator #40: [Request]:

kevidoplz, 2022-08-11:
### Please describe your request.

The ability to set joystick directions as 4 pin bindings so I can use 4 buttons to act as a left analog stick.

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-08-11:
so essentially your after the reverse of the "map left joystick to dpad" option?
new firmware should be able to do this but its a while away

kevidoplz, 2022-08-11:
> so essentially your after the inverse of the "map left joystick to dpad" option?
> new firmware should be able to do this but its a while away

Yeah, that’s actually pretty much exactly what I’m after

sanjay900, 2023-01-09:
Fyi this sort of thing is implemented in the new firmware,  i still don't have an ETA as there's an every growing list of things left to implement

closed 2023-01-09 by sanjay900