sanjay900/SantrollerConfigurator #41: [Issue]: Dpad left and right not working

aNOOBis98, 2022-08-13:
### Please describe your issue.

Hello, huge thanks for the tool, doin an awesome job, but i have a small issue, my dpad left and right doesn't seem to work.
I've cut the traces so the main board doesn get anything, wires are intact, no shortages at all, assigned dpad left to gp21 / dpad right to gp20.
Buttons work by themselves, i rang everything with a multimeter everything is good.
Tried to set dpad left/right as up/down and they actually work as up/down. Best regards, please help.
[image]

### Version

yes

### What operating system are you using?

Windows 10

### os-other

Win 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

aNOOBis98, 2022-08-13:
> ### Please describe your issue.
> Hello, huge thanks for the tool, doin an awesome job, but i have a small issue, my dpad left and right doesn't seem to work. I've cut the traces so the main board [...]

Yeah i broke the dpad down but i don't care, cuz it's on a strum anyway, so i cut it off

sanjay900, 2022-08-13:
Try turning off map joystick to dpad, it's probably intefering

aNOOBis98, 2022-08-13:
It's off

aNOOBis98, 2022-08-13:
Okay i am a pepega, it was on actually. My bad, my bad. Thank you very much. No one knows the stuff better than the author) wish you luck, have a good day

closed 2022-08-13 by aNOOBis98

sanjay900, 2022-08-13:
Im currently working on a new  firmware that should make it so that turning that on doesn't break the dpad outright, but that probably won't be ready for a while

aNOOBis98, 2022-08-13:
Wish you best of luck, you are doing a huge job for a huge community, keep it up!

bmasephol, 2023-02-27:
> Try turning off map joystick to dpad, it's probably intefering
> […](#)

This got my Left/Right working as well.  Keep up the good work, can't wait for the rewrite.  Let me know if you need a beta tester.