sanjay900/SantrollerConfigurator #42: [Issue]: During playing clone hero 1.0.0.xx usb device malfunctions

Murph9, 2022-08-17:
### Please describe your issue.

Sorry for the issue request, i tried quite a few things.

While successfully playing clone hero the Arduino pro micro stops responding and unplugging/re-plugging doesn't work.
Guitar was soldered and made last week with a new parts:
https://core-electronics.com.au/pro-micro-3-3v-8mhz.html
https://core-electronics.com.au/adafruit-wii-nunchuck-breakout-adapter-qwiic-stemma-qt.html
No tilt device installed.

Many attempts to connect the device to the computer later still malfunctions. With this message
[image]

It also doesn't show up in the found devices list when opening guitar configurator.

Clone hero restart, configurator restart and computer restart all didn't help.

The only thing i can think of is that I was playing the v0.23.2.2 clone hero release (from the website direct) and then installed the 1.x version through the discord downloadable launcher. Played at least 5 songs and then it failed as above.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 3.3V

### How is your controller programmed?

Wii

sanjay900, 2022-08-17:
That probably means your wiring is not sturdy enough.

Murph9, 2022-08-17:
Ah that sounds reasonable.
Well hopefully others can learn from this. 
Thanks

closed 2022-08-17 by Murph9

Murph9, 2022-08-17:
If i unplug the nunchuck adapter should it still work as a usb device?

sanjay900, 2022-08-18:
it should, its only because it gets stuck trying to read. Ill probably eventually put something in to make it time out when reading from a wii controller

Murph9, 2022-08-18:
I've double checked my soldering connections they all are properly convex [like](https://www.reddit.com/r/AskElectronics/comments/o4280e/are_these_solder_joints_good_enough/h2ey7c3/)

Is it possible a connection inside the guitar is the cause?

sanjay900, 2022-08-18:
I don't know what to tell you, it's always soldering when wii doesn't work
lol

Murph9, 2022-08-18:
The actual arduino (sparkfun) pro micro is bricked. I get no new com ports connections detected when doing the reset pin trick.
Symptoms are exactly the same as this:
https://www.reddit.com/r/olkb/comments/ghaiog/bricked_pro_micro_help_please/fq8tm6o/

Hes dead jim. https://www.youtube.com/watch?v=3at_Ev2kOoI

sanjay900, 2022-08-18:
get a pi pico for your next one man they are impossible to brick and far easier to recover if something does go wrong.

Murph9, 2022-09-14:
A follow up to the original issue, Pro Micro is still very much bricked - no fault of this tool.

A pi pico is much cheaper and has worked perfectly so far.