sanjay900/SantrollerConfigurator #44: [Request]: compatibility with ws2812 LEDs

nekoshade, 2022-09-06:
### Please describe your request.

The older apa102 LEDs are harder to obtain than the newer and better ws2812, also using only VCC, GND and data (one less pin)
It is possible to add a compatibility with those newer ones?

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-09-06:
WS2812s aren't better, as they require clocking data out at a very specific speed which causes issues when your dealing with interrupts. There's a reason i never supported them.

closed 2022-09-06 by sanjay900

nekoshade, 2022-09-08:
Thanks for the complex and detailed answer, everyone online was recommending ws2812 LEDs for compatibility and accessibility, so i questioned it not being here.
You clarified the reason and changed my mind about apa102, thanks again.