sanjay900/SantrollerConfigurator #45: [Request]: Option to calibrate Whammy Bar on a Wii connectivity

MrLuigi002, 2022-09-13:
### Please describe your request.

I transformed a Wiitar to a XInput guitar using your guide and a Raspberry Pi Pico a couple of days ago, and I've been testing it on a Xbox 360 using InvoxiPlayGames' UsbdSecPatch, which works perfectly. But I discovered that my guitar has a slight drift in the Whammy Bar, making it detected as slightly pressed all the time.

Some Guitar Hero games let you calibrate the Whammy, but others like Van Halen don't. So would it be possible to add it as an option in the GuitarConfigurator? I see you can do it if you do a Direct wiring, but not on a Wii or PS2 connection.

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2022-09-14:
unfortunately the calibration stuff is very tied into direct at the moment, theres a new firmware in the works that should let you do this but it is quite a way away yet.

sanjay900, 2022-12-09:
this has been implemented in the new firmware fyi, just still working on it still

MrLuigi002, 2022-12-09:
That's awesome, thank you!

sanjay900, 2023-03-22:
i know its been a while, but i am still working on this, just as an update here is a screenshot:
[image]

MrLuigi002, 2023-03-22:
I really appreciate the update, that config screen is looking amazing