sanjay900/SantrollerConfigurator #46: [Request]: Bind a button to multiple pins

stevenla, 2022-10-16:
### Please describe your request.

Sometimes tilt activation is hard to set up properly (e.g. RPCSX3 Rock Band is kind of finicky), so it'd be nice to be able to bind digital tilt pin to select IN ADDITION to having the select button already bound to a pin. Kind of like the inverse of #22. This would be a lot easier for users than opening the controller up and shorting tilt to select.

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-10-16:
new firmware will let you do this sort of thing, can't give you an eta

stevenla, 2022-10-17:
take your time, you've made an amazing impact on GH controller modding already :)

sanjay900, 2023-03-22:
I did do some tilt changes in the latest version of the config tool, should hopefully make tilt a bit better for PS3 guitars in GH mode on console, if it helps let me know