sanjay900/SantrollerConfigurator #47: [Request]: Player 1-4 Led support

gitmesomepizza, 2022-10-19:
### Please describe your request.

I would like to restore the ability to use the LEDs that are pre-built into some guitars to indicate if a guitar is player/controller 1, 2, 3, or 4.
Xbox guitars had the Xbox 4 LED ring, PlayStation had the 4 red LEDs. Those LEDs should already be rated around 3v.

Also, thank you for making this awesome tool!

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-10-19:
Its on the list, but the current version of the firmware has some technical limitations that prevent this from being possible. Theres a new firmware in the works that would make it possible to implement this but its still being worked on at the moment.

ujicos, 2023-03-05:
This is a dirty workaround, but what i did with my Arduino Pro Micro was run a wire from the power led on the arduino to the P1 led on the guitar

sanjay900, 2023-03-05:
that works but its no player led haha

ujicos, 2023-03-05:
close enough :troll:

sanjay900, 2023-03-05:
I probably should mention that the support is coded into the new firmware since i never did update this