sanjay900/SantrollerConfigurator #48: [Issue]: guitar keeps "crashing"

redstone47, 2022-10-21:
### Please describe your issue.

the guitar will randomly stop recognizing any button inputs and will continue holding down notes that where held down before. i also had another uno modded guitar do this and the only fix was to make a reset button for the arduino. 

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino uno

### How is your controller programmed?

Direct

sanjay900, 2022-10-22:
I guess the first thing i can say is don't use  unos if you can, use picos.
I'm working on a new firmware that will hopefully help with stuff like this but out of curiousity what pins are you using on the uno?

redstone47, 2022-10-22:
i would have used a pico but i didnt have any and i already had the uno and the pin bindings are 6, 7, 4, 3, 12, 10, 11, 9, 8, and A0 for whammy.

sanjay900, 2022-10-27:
One thing that you could possibly try is seeing if any of the older versions have this bug fixed?

redstone47, 2022-10-27:
it has the problem on the old version too

sanjay900, 2022-10-27:
its interesting right cause i know a lot of people with uno guitars that never had this issue, which uno revision have you got?

redstone47, 2022-10-27:
its some random off brand of the rev 3 i think

sanjay900, 2023-05-07:
@redstone47 i know its been a while, but are any of the newer versions working better for you?

redstone47, 2023-06-02:
yes, the newer versions have been working alot better and i havent had any issues sence

closed 2023-06-02 by sanjay900