sanjay900/SantrollerConfigurator #49: [Issue]: Not really an issue, just wondering about powerstar led

jackw163, 2022-10-22:
### Please describe your issue.

how should the powerstar led be wired in? 

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2022-10-23:
youd get the correct current limiting resistor for your led, there are calculators for that on the internet
the pin will provide 5v when star power is active, so you just wire the led and resistor between the pin and ground

jackw163, 2022-10-23:
Perfect, i thought it would be like that but with nothing about it on the wiki i just wanted to make sure, Thak you for your reply and Thank you for this software its amazing!

sanjay900, 2022-10-23:
Heh the feature only exists cause someone asked for it, i haven't really documented much of the LED stuff as it only works with CH v23 and im trying to find a better way to go about it

closed 2023-02-10 by sanjay900