sanjay900/SantrollerConfigurator #50: [Request]: Led Brightness

jackw163, 2022-10-23:
### Please describe your request.

Hi, 

I realise this could be permanently achieved via hardware, but could a led brightness slider be added for the led options in the guitar configurator? would then make it easy to adjust all led brightness's on the fly especially as you may want the Starpower a different brightness to the fret Leds or something.

Many Thanks
Jack

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

arduino pro micro 5V

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-10-24:
Are you talking about APA102s or normal leds?

jackw163, 2022-10-24:
Both, but mostly normal leds, or even just a bright and dim setting

sanjay900, 2022-10-24:
normal leds i can't really do anything about, as the microcontroller itself doesn't actually output anything for driving leds behind frets. You would have to change your resistor value to change brightness.

jackw163, 2022-10-24:
No worries, As a side note, could APA102's also be used for lighting for
other parts of the guitar? or even just ambient lighting?

sanjay900, 2022-10-24:
you can assign them to any button, i haven't really got any code for ambient stuff, but it would probably make more sense to drive them with another microcontroller if they are just always going to be the same colour

jackw163, 2022-10-24:
Awesome, thanks for your replies they've really helped me plan my setup

closed 2023-01-11 by sanjay900