sanjay900/SantrollerConfigurator #52: [Issue]: Tilt Sensor is incorrect?

NightmareIris, 2022-11-16:
### Please describe your issue.

Unsure if I set my sensor wrong to begin with, but for whatever reason. My tilt is acting similar to an official guitar, in my case an Xplorer, but always triggers starpower on consoles. 

I used SW-520D sensors, which seem to only allow two positions, either on or off. Im guessing this is by design of said switch. But, they never seem to sit at complete 0. It'll sit at its resting angle of -0.00002, which I believe is causing consoles to think its being tilted constantly. Unlike my Xplorer which rests at 0 and goes below that. My sensor is set to "digital". Is this a me issue or the the programs issue?

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2022-11-16:
which console?

NightmareIris, 2022-11-16:
Xbox 360, specifically used on Warriors of Rock and Rock Band 3.

Warriors of Rock would always go instantly. Rock Band 3 would either wait or go instantly.

sanjay900, 2022-11-17:
It would not surprise me to learn that the 360 stuff tilt doesn't work since that wasn't even a supported thing until pretty recently

NightmareIris, 2022-11-17:
Doesn't surprise me either, but would there be a way to "fake" the sensors resting angle? Since Im gonna guess the games see it's -0.00002 angle as activated.

sanjay900, 2022-11-17:
It's odd because I'm not sure why it wouldn't rest at 0 with the current code, I'll have a look later

NightmareIris, 2022-11-17:
If it matters I do only have 1 sensor on the board, but it seems to be fine outside of console. So if you can figure it out, I'd be greatly appreciative.