sanjay900/SantrollerConfigurator #53: [Request]: Support for Teensy boards

y2k04, 2022-11-22:
### Please describe your request.

It would make my life a lot easier if this program was compatible with the Teensy. I have the 4.0 version.

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-11-23:
It's in the backlog but it's not as easy as you might think, and it's something that will come after a rewrite.

y2k04, 2022-12-01:
Is there any documentation or comments in your code? Wondering whether it would be possible to try and contribute it, or at least attempt to. I have already made my own code for the Teensy 4.0 and works pretty good. Although, the only thing I tested was the only soldered down fret button, which was the green, and it was solid.

y2k04, 2022-12-01:
Also if it helps, the pin out looks similar to the Arduino Pro Micro

sanjay900, 2022-12-01:
> Is there any documentation or comments in your code? Wondering whether it would be possible to try and contribute it, or at least attempt to. I have already made my own code for the Teensy 4.0 and [...]

Can't really recommend doing that since the codebase is going through a complete rewrite at the moment.

sanjay900, 2022-12-01:
The other problem is the teensy seems to require modifying the teensy libs themselves to get access to usb at the level needed for this project to function

sanjay900, 2022-12-02:
i do have half a teensy implementation sitting there in the new codebase (like all the building and stuff) but i have other priorities at the moment before i can finish that.

y2k04, 2022-12-02:
Okay :)