sanjay900/SantrollerConfigurator #54: [Issue]: incorrect Drum mapping

JR2703, 2022-11-28:
### Please describe your issue.

When having the connectivity type set to Wii and the output to PS3 Guitar hero drums the mapping is incorrect. The yellow cymbal is set to Blue drum & the kick pedal is set to the orange cymbal. 

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Wii

sanjay900, 2022-11-28:
Ugh, why can't they just be consistant with things haha, ill have to sort this later

sanjay900, 2022-12-08:
Ive got a drum implementation in the new firmware, but can't give you an ETA for that.

JR2703, 2022-12-08:
Appreciate it. Thank you!

RetroCult, 2022-12-30:
Just thought I'd throw in that it appears that the output is also incorrectly mapped in the same way for every configuration of "Guitar Hero Drums".

Looking forward to the new firmware!

sanjay900, 2023-01-01:
yeah, ive got that all fixed now in the new firmware, turns out the guitar hero guitars are the only ones that flip the pads, e very other controller uses it the other way around

Cicabe, 2023-06-25:
Hey, did this ever actually get fixed? I downloaded the version that just released earlier today (for Mac if that helps) and I'm experiencing this issue. Trying to configure a Wii Guitar Hero (5/Band Hero) drum kit for PS3 and it's swapping yellow with blue and kick with orange. There doesn't seem to be a way to fix it. No remapping options or anything.

sanjay900, 2023-06-25:
its fixed in a testing firmware but not the normal one

Cicabe, 2023-06-25:
Thanks for the reply. Any chance I could get the testing firmware?

sanjay900, 2023-06-25:
a 9.4.3 should come out soon that has them flipped the right way around

Cicabe, 2023-06-25:
Good to know! I'll keep an eye out for it.

Cicabe, 2023-06-25:
Okay so the mapping is right, even though GC shows it as incorrect. But I also feel like I need to hit the drums really hard to get them to register in game? Like, harder than I'm comfortable hitting them. Even at the lowest drum threshold, it doesn't seem to register consistently. Is threshold supposed to be for the sensitivity, or is that something else?

sanjay900, 2023-06-25:
the threshold stuff is for direct wiring, weirdly im just using what the drums themselves send me, as they send a digital signal when hit.

sanjay900, 2023-06-25:
Activision did provide an application for adjusting the sensitivities via MIDI, https://archive.org/details/redoctane-drum-tuner

Cicabe, 2023-06-25:
Guess this application would be my best bet for trying to get this thing to work. Sadly it doesn't seem to recognize the Ardwiino, and a MIDI to USB cable is a bit pricier than I'm comfortable shelling out for at the moment, especially for something that's not guaranteed to work. Guess I'll put this on the backburner for now.

sanjay900, 2023-06-25:
If you have another arduino, you might be able to build a midi to usb cable, but unfortunately i don't see a way to adjust the sensitivies from my end

Cicabe, 2023-06-25:
Okay the drums actually work perfectly fine playing in Clone Hero when configured as PS3 drums. So now I think this is a problem on the PlayStation end of things.

sanjay900, 2023-06-27:
Bizarre, that means the PS3 is specifically looking at the velocities which is annoying