sanjay900/SantrollerConfigurator #55: [Request]:

Dark1886, 2022-11-28:
### Please describe your request.

Requesting support for ADAFRUIT QT PY - SAMD21 board. 

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2022-11-30:
I may look into this one day, but i don't own one of these, so its not something i can do easily.

Dark1886, 2022-11-30:
I'll be getting one very soon and may see if I can manage some updates to
get it working, but I'll admit the depth of this project is a little
outside my typical software workloads.

sanjay900, 2022-11-30:
Yeah I wouldn't recommend that with the current firmware it's very tied to
the Pico and the arduino

Dark1886, 2022-12-03:
> Yeah I wouldn't recommend that with the current firmware it's very tied to the Pico and the arduino. Theres a rewrite on the way, and until that is done i would hold of editing anything, but once thats there it uses platform.io so things should be easier.
> […](#)

I’ve heard about a re-write, and I’m wondering when the ETA (Roughly) on that is? Sorry If i missed this mentioned somewhere.

sanjay900, 2022-12-04:
Heh, theres no ETA

Dark1886, 2022-12-05:
> Heh, theres no ETA, hence why you haven't seen one

No problem, if it was around the corner I would wait. I'll probably just end up with a different controller to use instead.

closed 2022-12-05 by Dark1886