sanjay900/SantrollerConfigurator #56: [Issue]: Stuck on programming USB controller

kizzoknight, 2022-11-30:
### Please describe your issue.

using a arduino pro micro 3.3v gets to programming usb controller (%100) then doesnt do anything

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 3.3V

### How is your controller programmed?

Wii

sanjay900, 2022-11-30:
What's your computer see the device as after this point

kizzoknight, 2022-11-30:
Apologise for some reason I needed to select 5v even tho I did it
previously on a board that was exact same that worked with 3.3v

sanjay900, 2022-12-01:
Theres two different variants of the board, you must have brought a 3.3v one previously and a 5v one the second time, they look identical but have a single changed component hence you probably wouldn't have noticed the difference

closed 2022-12-01 by sanjay900