sanjay900/SantrollerConfigurator #57: [Request]: Support for Seeed XIAO RP2040

kizzoknight, 2022-12-03:
### Please describe your request.

https://wiki.seeedstudio.com/XIAO-RP2040/

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-12-04:
What does it do when you try to use it?

kizzoknight, 2022-12-04:
I've tried it as a qt py 2040 as that's the closest board on there and it uploads fine but no input when testing it

kizzoknight, 2022-12-04:
I think I tried that. I can give it another try tomorrow and also send some pics of how it's all wired

kizzoknight, 2022-12-04:
Cool I'll send some pics tomorrow

sanjay900, 2022-12-04:
Im going to assume this is probably an issue due to the fact that the RP2040 supports multiple different pins for I2C, and this board is probably using different ones than the ones that i have picked. Theres a new firmware on the way that will fix this but for now you wont be able to do wii stuff with this board

sanjay900, 2022-12-04:
Yep, just had a look and can confirm that is the problem

sanjay900, 2022-12-07:
I probably should mention, theres a new firmware in the works to support exactly this, but there is not ETA at the moment