sanjay900/SantrollerConfigurator #58: [Request]: LED's

kizzoknight, 2022-12-07:
### Please describe your request.

Multiple requests in one but when using 5 APA102 LED's would it be possible to get the whammy bar to change the brightness of them when whammying

Pressing star power button can flash the standard LED's a colour of users choice

When in star power have the ability to change colour profile so you can have 5 diffrent colours when in star power

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-12-08:
>Multiple requests in one but when using 5 APA102 LED's would it be possible to get the whammy bar to change the brightness of them when whammying
already implemented in a new firmware thats in development
>Pressing star power button can flash the standard LED's a colour of users choice
this would also be possible there too
>When in star power have the ability to change colour profile so you can have 5 diffrent colours when in star power
This sort of thing may be possible, i have not yet implemented star power stuff in the new firmware yet.

kizzoknight, 2022-12-08:
Cool you dont have a ETA on the new firmware do you?

kizzoknight, 2022-12-10:
Sorry would pressing the star power button be able to flash the fret APA102's led's white or another colour of users choice

sanjay900, 2022-12-10:
yes

closed 2022-12-10 by kizzoknight