sanjay900/SantrollerConfigurator #60: [Issue]: Tilt Sensor

kizzoknight, 2022-12-13:
### Please describe your issue.

When I have a tilt sensor connected its always active when i chage the setting of the sensitvity in clone hero it just doesnt work at all

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Wii

sanjay900, 2022-12-13:
Are you using PS3 mode or xinput

kizzoknight, 2022-12-13:
xinput

kizzoknight, 2022-12-13:
it also shows on z rotation but arent tilt switches either on or off so should just be a button press right?

sanjay900, 2022-12-13:
Nah it should map tilt to an axis, because there what the game expects

You shouldn't need to calibrate anything on clone heros end

sanjay900, 2022-12-13:
what is it showing on z rotation when you tilt?

kizzoknight, 2022-12-13:
[image]
Z rotation is what changed when tilting

sanjay900, 2022-12-13:
and how is it changing exactly? sounds more like you just need to reset the calibration settings in ch

sanjay900, 2022-12-13:
also, how do you have your tilt sensor connected

kizzoknight, 2022-12-14:
ive tried the tilt sensor between GND and an Analogue pin and GND and a digital pin and it jumps from half way on Z rotation to full i will try the calibration in windows soon

sanjay900, 2022-12-14:
did you tell the tool you had an analog sensor?

kizzoknight, 2022-12-14:
yes and selected the pin

sanjay900, 2022-12-14:
try hitting the "invert" button inside of clone hero

kizzoknight, 2022-12-14:
that makes it not work at all

sanjay900, 2022-12-14:
so hold on what is it doing now, is it just always active
and then inverting makes it never activate?

ruenes1, 2023-06-19:
Hello,

I think I'm having the same issue. I use a digital tilt sensor SW-520D and, when it is not titled, the Z rotation axis is all the way up and CH interprets is a start power. Only when I tilt it, then the Z rotation axis is neutral. I'm sure the orientation of the physicial tilt sensor is correct because I connected a led in line, and it behaves as expected (and inversely to the expected behavior of the Z rotation axis). Using the setting to invert tilt into the configuration tool does nothing (as I understand it is ment for analog tilting).

Thanks

sanjay900, 2023-06-19:
Have you set it to digital tilt in the tool?

ruenes1, 2023-06-20:
Yes I've set it to digital. But I think I found the issue. Since I've wired a led in line with the switch by following the schematic for the frets, the input pin is receiving vcc rather than gnd when the sensor is tilted, thereby resulting in inversed behavior. Now I've wired it like a standard button without the led and it works as expected. Having the inline led was nice to check when the titling actives/deactivates but I understand the current firmware only supports leds for frets.

sanjay900, 2023-06-21:
Ah yeah that would do it

closed 2023-07-14 by sanjay900