sanjay900/SantrollerConfigurator #62: [Request]: Pro Drums support for Rock Band

rjkiv, 2022-12-19:
### Please describe your request.

The latest version of the RB Drum configurator tool does not appear to have support for pro drums (i.e. supporting the yellow cymbal, blue cymbal and green cymbal). Adding functionality for these cymbals would be great - thank you for reading this issue and for your hard work on this!

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

arduino micro 5V

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-12-19:
support is in progress in a new firmware thats on the horizon, cant give you an ETA unfortuantely

rjkiv, 2023-01-01:
I just realized, for straight up pro drums, you would need an analog pin for each pad (so seven in total for pro). I'd like to ask, would there be a possible option in the future to use your configurator as a mock MPA (i.e., MIDI e-kit to an Arduino/Pico/etc)? This way, every input into your Arduino/Pico/etc would already be digitalized, thus not needing to have a board with 7 analog pins, or having to do some crazy threaded logic in order to reserve analog pins for specific drum hit combos (i.e. two cymbals and a kick)

sanjay900, 2023-01-01:
im planning on just adding support for analog multiplexer chips to solve that

rjkiv, 2023-01-01:
Gotcha, figured I'd ask anyway - thank you for the quick response and for your hard work!

eckerj, 2023-03-17:
Hello, I'm also very interested in analog multiplexer support for the 2040.  Is there anything I can do to help?  I have a few 2040s as well as some 16 channel CD74HC4067 analog multiplexers on hand.

sanjay900, 2023-03-17:
I've got multiplexers as well, it's something on the list

sanjay900, 2023-03-19:
> Hello, I'm also very interested in analog multiplexer support for the 2040. Is there anything I can do to help? I have a few 2040s as well as some 16 channel CD74HC4067 analog multiplexers on hand.

[image]

eckerj, 2023-03-19:
Oh wow, that's looking perfect! Especially the calibration portion.  I'm planning on using potentiometers in my drum pad design and being able to see the input values real-time is going to be awesome.

eckerj, 2023-03-19:
Does this mean you're close to including Pro Drums support (including velocity)?

Thank you for all your hard work, Sanjay!

sanjay900, 2023-03-19:
It's pretty much all implemented yeah