sanjay900/SantrollerConfigurator #63: [Request]: Pickup selector

kizzoknight, 2022-12-21:
### Please describe your request.

Could you assign the pickup selector potentiometer to an axis for the rock band guitar so we can set as whatever we want in game

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2022-12-21:
You already can do this, you just need to set your guitar as a rockband guitar not a guitar hero one.

kizzoknight, 2022-12-21:
yeah its set as rockband guitar and configured in the software and working but doesnt show as an axis in game or in windows

sanjay900, 2022-12-21:
are you using PS3 mode or xinput

kizzoknight, 2022-12-21:
xinput

kizzoknight, 2022-12-21:
Uploading Guitar Configurator 2022-12-21 01-36-56.mp4…

kizzoknight, 2022-12-21:
https://youtu.be/UXuJ1cqD4ic

This slider moving was just me leaning on the whammy

sanjay900, 2022-12-21:
swap your ground and vcc wires

kizzoknight, 2022-12-21:
[image]
thats not worked im afraid ive tried it with min and max both ways round and inverted and not inverted

sanjay900, 2022-12-21:
No idea what's going on then you'll just have to wait for the new firmware

sanjay900, 2023-04-13:
So i read this again, and i think i might actually know what the issue is here
I don't think xinput on windows understands how the slider bar or pickup selector works.

I suspect if you use PS3 mode, the pickup selector will just show up and be usable.