sanjay900/SantrollerConfigurator #64: [Issue]: PS2 controller always has some Whammy even when not pressed.

rwillsuk, 2022-12-21:
### Please describe your issue.

Looks like my PS2 SG is showing that there is always some whammy being pressed, even when I'm not pressing it. Not sure if its a guitar issue or an adapter issue (built your adapter with a pico). Would be nice if it could be calibrated.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

PS2

sanjay900, 2022-12-22:
This will be possible in the future after i release a rewrite thats in progress

elgarlic, 2023-01-15:
I just wanted to chime in that I am also experiencing this issue, and my setup details:
v9.1.2. PS2 input, PS3 Rockband Guitar output. Playing Rockband 4 on PS5. Pi Pico. 

Probably to be expected, but programming it for Guitar Hero guitar output results in no whammy signal whether pressed or not.

sanjay900, 2023-04-16:
Would you be able to get me the value that the whammy sits  at from a controller tester? I can also code in a sensible default so that you don't even need to calibrate in most cases in the new firmware.

MicrocontrollersDev, 2023-06-05:
Not the original sender but since they haven't responded thought I'd add my own. My whammy bar sits at -0.43529 when not pressed on a GH5 Wiitar

sanjay900, 2023-06-06:
> Yep, tho I'm having this issue with adding you
> [image: DiscordCanary_BDjTkW8ESq]
> <https://user-images.githubusercontent.com/66657148/243284236-29210dd3-f598-4d63-8107-e578c18b8994.png>
>
> This might be on me I have no clue how discord's new username system works
> so

Sorry yeah, at some point my nitro expired, sanjay900#2732

MicrocontrollersDev, 2023-06-06:
Yep, worked, sent the friend req

sanjay900, 2023-07-05:
> I just wanted to chime in that I am also experiencing this issue, and my setup details: v9.1.2. PS2 input, PS3 Rockband Guitar output. Playing Rockband 4 on PS5. Pi Pico.
> 
> Probably to be expected, but programming it for Guitar Hero guitar output results in no whammy signal whether pressed or not.

funnily enough this might be a different issue from what ive learned recently. The PS3 games just handle whammy totally differently to how i expected, i will have to implement that differently at some point.