sanjay900/SantrollerConfigurator #66: [Issue]: Wiring Whammy with only 2 wires.

bmasephol, 2023-01-02:
### Please describe your issue.

I'm working on converting a PS3 Fender Wireless Precision Bass (model 9766) because the USB dongle is long gone.  

This device only has 2 wires for the Whammy.  When I inspected the rotary switch and PCB its connected to I found that 2 of the 3 connections for the rotary switch are soldered together and then just 2 wires run back to the main PCB.  Those are labeled Whammy and AD, (I think AD on this board is the positive power) as there are plenty of other spots labeled GND for ground.  The wire that is connected to the 2 bridged spots is connected to the Whammy spot on the main PCB.

Just looking for ideas on what you think and if it will work.

[image]

[image]

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2023-01-02:
That is... Odd. Tbh you could probably just desolder that join and bring out a third wire and do it that way.

bmasephol, 2023-01-05:
I've been doing some more research on this setup, I think it will work.  Essentially they implemented the whammy bar in this guitar model using a rheostat instead of a potentiometer.  The interesting thing I've learned is that you can make a rheostat out of a potentiometer by doing exactly what is pictured, connecting the group pin to the signal pin.

I think that the rheostat was chosen as it controls the resistance in a more definitive way of either being 0 ohms or max ohms when the wiper (dial) is turned all the way.  I plan to do some more studying on this.

A follow up question is since this is connected to an analog pin on the pi pico, you must be monitoring the voltage drop to determine if the whammy bar is activated or not... is that on the correct track?

sanjay900, 2023-01-05:
Give it a go it might just work, if it doesn't you may just need to put a
resistor between the analog pin and ground

sanjay900, 2023-01-05:
I have seen people short a pin on a potentiometer before as well to run it from two pins

bmasephol, 2023-01-07:
Just to follow up with a bit more data on this as I've been digging and learning.  The potentiometer in play here is a B103, something like this https://www.mouser.com/ProductDetail/Bourns/PTV09A-4020F-B103?qs=Qzws7J6gxqx9VaKCiVoniw%3D%3D

So a 10k ohm pot, but wired up as a rheostat.  I'm going to pull some readings off it with my multi meter later this weekend and see what the range will be with 3.3v as the source power voltage.

Does the code activate the whammy action with full voltage or less voltage?

sanjay900, 2023-01-07:
If it's a 10 K pot, your best bet is to just desolder that bridge and
solder your own three wires. You can calibrate it, but pots do tend to work
better when all three pins are used

bmasephol, 2023-01-08:
Ok, so I'm doing a complete rewire after my first attempt.  Good news is the whammy bar was recgonized and I think I was able to calibrate it leaving it as a rheostat.  I just didn't like how my wiring worked the first time around and I had some issues with some controls because of how the original guitar PCB did things.

I do have a question about the D-pad and the double strum that is on the guitar I'm doing.  Are the Down and Up on the D-pad, the strum up and strum down technically the same thing, and should be all wired together?  Ex: D-pad down, 2x strum down should all connect to say GP10?

sanjay900, 2023-01-08:
yeah thats exactly how to wire the dpad

closed 2023-01-10 by sanjay900