sanjay900/SantrollerConfigurator #69: [Issue]: Guide button presses frantically, practically useless

jwagriff, 2023-01-23:
### Please describe your issue.

Just built a direct guitar, an am trying to bound a button to the guide button in the configuration. The button does get binder, but once it does, looking at game pad tester, the button is pressed for very short amounts of time despite being held down continuously/spam pressed.

it’s not a hardware problem, as I can re-bind the same pin to start and I have no issues there.

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2023-01-24:
turn off map start + select to home, i suspect thats conflicting

sanjay900, 2023-03-16:
Did this work?

sanjay900, 2023-03-29:
I can confirm the guide button has no issues in the new firmware in development as i use it a lot when testing console stuff at the moment.

sanjay900, 2023-04-10:
Did some rewrites of map start + select to home, it shouldnt flicker now but if you have it enabled then you will just find that you wont get any inputs from a physical guide button as both of these options can't be used at the same  time.

closed 2023-04-10 by sanjay900